Robo4J/robo4j #34: Move to Pi4J 2.0

thegreystone, 2019-09-05:
Now that Pi4J 2.0 is starting to contain all that is needed for Robo4J, we should start adopting it. Note that we shouldn't officially adopt it until the first version has been properly released.

java-boy, 2021-12-23:
Hi @thegreystone 
it seems that Robo4j is not being developed anymore.
Shall we use Pi4j instead of Robo4j in our projects?
Does Pi4j support all functionalities and libraries available in Robo4j?

java-boy, 2021-12-23:
Do you have any plan to update your Cofee robot to use Pi4j ?

mirage22, 2021-12-24:
Hi @java-boy , we have been already running Robo4j on Pi4j early releases during CodeOne. Sadly we ware not able to allocate properly a time to migrate main branch to Pi4j 2.0 as it was not finished that time.  I may have good feeling that migration may happen early next year, Maybe even with migration to Java17+ as I've been already playing with it.

java-boy, 2021-12-25:
Thanks, @mirage22, so Robo4j is already running on Pi4j. can't wait to see your new update on Robo4j to run on Pi4j 2.0
Best wishes.

thegreystone, 2021-12-27:
Hey there! Sorry for the long hiatus. Too much has been happening lately, and I've simply not had time for Robo4J. Yes, perhaps this upcoming year would be a good time to dust off the robotics hobby and get going again. I have a pretty exciting Coff-E 2 design that would be fun to complete and share the blueprints for.

java-boy, 2021-12-28:
Great news @thegreystone 
Waiting for new exciting updates.

mirage22, 2023-10-17:
Currently robo4j uses a self compiled old v2.0-SNAPSHOT version and should be replaced by most recent version 2.3.0

closed 2023-11-06 by mirage22